woglinde/meta-java #81: openjdk-7-jre build error on dora branch

mirzak, 2015-01-07:
Compiling worked fine a few days ago but now I got below error. 

This error is also mentioned at http://mail-index.netbsd.org/pkgsrc-users/2014/12/31/msg020849.html.

It is clear that some hardcoded date is producing this error. As in the mail discussion above there is an hack to fix this but wanted to post here and see what you guys think.

```
| /media/sdb1/oe-core-new/build/out-eglibc/work/armv7ahf-vfp-angstrom-linux-gnueabi/openjdk-7-jre/25b30-2.3.12-r5.1/icedtea-2.3.12/build/bootstrap/jdk1.6.0/bin/java -XX:-PrintVMOptions -XX:+UnlockDiagnosticVMOptions -XX:-LogVMOutput -client -Xmx1536m -Xms1536m -XX:PermSize=32m -XX:MaxPermSize=160m -jar /media/sdb1/oe-core-new/build/out-eglibc/work/armv7ahf-vfp-angstrom-linux-gnueabi/openjdk-7-jre/25b30-2.3.12-r5.1/icedtea-2.3.12/build/openjdk.build/btjars/generatecurrencydata.jar -o /media/sdb1/oe-core-new/build/out-eglibc/work/armv7ahf-vfp-angstrom-linux-gnueabi/openjdk-7-jre/25b30-2.3.12-r5.1/icedtea-2.3.12/build/openjdk.build/lib/currency.data.temp \
| 		< ../../../src/share/classes/java/util/CurrencyData.properties
| Error: time is more than 10 years from present: 1104530400000
| java.lang.RuntimeException: time is more than 10 years from present: 1104530400000
| 	at build.tools.generatecurrencydata.GenerateCurrencyData.makeSpecialCaseEntry(GenerateCurrencyData.java:285)
| 	at build.tools.generatecurrencydata.GenerateCurrencyData.buildMainAndSpecialCaseTables(GenerateCurrencyData.java:225)
| 	at build.tools.generatecurrencydata.GenerateCurrencyData.main(GenerateCurrencyData.java:154)
| make[5]: *** [/media/sdb1/oe-core-new/build/out-eglibc/work/armv7ahf-vfp-angstrom-linux-gnueabi/openjdk-7-jre/25b30-2.3.12-r5.1/icedtea-2.3.12/build/openjdk.build/lib/currency.data] Error 1
| make[5]: Leaving directory `/media/sdb1/oe-core-new/build/out-eglibc/work/armv7ahf-vfp-angstrom-linux-gnueabi/openjdk-7-jre/25b30-2.3.12-r5.1/icedtea-2.3.12/build/openjdk/jdk/make/java/java'
| make[4]: *** [all] Error 1
| make[4]: Leaving directory `/media/sdb1/oe-core-new/build/out-eglibc/work/armv7ahf-vfp-angstrom-linux-gnueabi/openjdk-7-jre/25b30-2.3.12-r5.1/icedtea-2.3.12/build/openjdk/jdk/make/java'
| make[3]: *** [all] Error 1
| make[3]: Leaving directory `/media/sdb1/oe-core-new/build/out-eglibc/work/armv7ahf-vfp-angstrom-linux-gnueabi/openjdk-7-jre/25b30-2.3.12-r5.1/icedtea-2.3.12/build/openjdk/jdk/make'
| make[2]: *** [jdk-build] Error 2
| make[2]: Leaving directory `/media/sdb1/oe-core-new/build/out-eglibc/work/armv7ahf-vfp-angstrom-linux-gnueabi/openjdk-7-jre/25b30-2.3.12-r5.1/icedtea-2.3.12/build/openjdk'
| make[1]: *** [build_product_image] Error 2
| make[1]: Leaving directory `/media/sdb1/oe-core-new/build/out-eglibc/work/armv7ahf-vfp-angstrom-linux-gnueabi/openjdk-7-jre/25b30-2.3.12-r5.1/icedtea-2.3.12/build/openjdk'
| make: *** [stamps/icedtea.stamp] Error 2
| ERROR: oe_runmake failed
| ERROR: Function failed: do_compile (see /media/sdb1/oe-core-new/build/out-eglibc/work/armv7ahf-vfp-angstrom-linux-gnueabi/openjdk-7-jre/25b30-2.3.12-r5.1/temp/log.do_compile.31112 for further information)
NOTE: recipe openjdk-7-jre-25b30-2.3.12-r5.1: task do_compile: Failed
ERROR: Task 5296 (/media/sdb1/oe-core-new/build/../stuff/meta-java/recipes-core/openjdk/openjdk-7_25b30-2.3.12.bb, do_compile) failed with exit code '1'
NOTE: recipe libmad-0.15.1b-r3: task do_package: Succeeded
NOTE: recipe mpeg2dec-0.4.1-r2: task do_package: Succeeded
NOTE: recipe classpath-0.99-r3: task do_compile: Succeeded
NOTE: Tasks Summary: Attempted 6597 tasks of which 3155 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /media/sdb1/oe-core-new/build/../stuff/meta-java/recipes-core/openjdk/openjdk-7_25b30-2.3.12.bb, do_compile
Summary: There were 6 WARNING messages shown.
Summary: There was 1 ERROR message shown, returning a non-zero exit code.
Command returned error code 1. Exiting
Command returned error code 1. Exiting
Cleaning up...
Build step 'Execute shell' marked build as failure
Archiving artifacts
Finished: FAILURE
```

mirzak, 2015-01-07:
Also found this patch http://icedtea.classpath.org/hg/release/icedtea7-forest-2.5/jdk/rev/9afae42eae54

But do not really know how to backport that in good way.

woglinde, 2015-01-07:
Instead of backporting stuff we should use newer release, unfornatly I do not have the time to do this and test it properly

mirzak, 2015-01-07:
Even in dora branch? I can understand that we do not want to backport in master branch

woglinde, 2015-01-07:
Even in dora or do you want to backport all the security related stuff?

wrotte, 2015-02-09:
Would it be possible to get a patch of some kind?  For example:

http://lists.openembedded.org/pipermail/openembedded-devel/2014-December/099545.html

otavio, 2015-02-09:
Please use the meta-java from Yocto Project git server; it has the patches merged.

wrotte, 2015-02-09:
Ahh, thanks.  Didn't realize yocto was maintaining a fork.  Good to know.

mirzak, 2015-02-09:
Yes that was the best solution

Malek44, 2015-03-26:
As a workaround to this problem, roll back your clock to 2014. This should work.
Details: https://bugs.gentoo.org/show_bug.cgi?id=534118

woglinde, 2016-01-13:
meta-java has moved you can find it now under

git://git.yoctoproject.org/meta-java

closed 2016-01-13 by woglinde